SUSTAINA-OP/SUSTAINA-OP #351: SUSTAINA-OP Ver. 2.0の切削記録

MasatoKubotera, 2023-03-07:


MasatoKubotera, 2023-03-07:
@KiyoshiroKawanabe 

3/10に以下の部品を切削お願いします．

名称 | 材料 | 数量
-- | -- | --
Trunk_neck_frame | A2017 T4 | 6
Trunk_shoulder_frame  | A2017 T4 | 6
Trunk_shoulder_frame_MIR  | A2017 T4 | 6

KiyoshiroKawanabe, 2023-03-09:
## CAMの名前
A2017_T4_01
A2017_T4_0
## 担当者名
@KiyoshiroKawanabe 
## commit ID
ed312e65bff4c41cb8f24517bc7e94d5e1e4669c
079f165350e8ecb16c00c48c256ba59dde08aeca 

## 配置したパーツと個数
Trunk_neck_frame ×6
Trunk_shoulder_frame ×6
Trunk_shoulder_frame_MIR ×6

## 切削所要時間
3:21:20
1:57:39

## 配置後のCAM画像
[image]
[image]

KiyoshiroKawanabe, 2023-03-10:
# 切削報告
## 切削日
2023/03/10
## CAM
A2017_T4_01
## 板厚公差
[image]
右上から時計回りに，
+0.01, +0.02, +0.02, +0.01

## 切削完了
[image]

## エンドミル
アルミ特化の01番を使用
[image]
## 加工距離

84.3m

KiyoshiroKawanabe, 2023-03-16:
# 切削報告
## 切削日
2023/03/13
## CAM
A2017_T4_02
## 板厚公差
[image]
全て+0.05mm

## 切削完了
[image]

## エンドミル
アルミ特化の01番を使用
かなり高いビビり音が鳴っておりエンドミルの限界を感じた．
今回の加工距離は50.2ｍで，01番の総加工距離は134.5ｍである．
今回の切削はすべて速いパラメータで行ったので，遅くすればもっと寿命は延びると考えられる．
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-04-28:
## 切削メンバーの作業確認

@RikuYokoo @AD58-3104 @MibuchiYuta でGFRP T2 (Green) の廃材を用いて足裏のパーツを切削し、今後切削できるかを確認しました。
12号館4階 学生自由工作室のCNCで作業を行いました。

### 1日目   
作業動画 [230424 切削お試し 井上三渕横尾](https://studio.youtube.com/video/4Qhop6fqrD8/edit?c=UC1AFpP2tjB5Kh3M82s_2XFA)
CAMデータにミスがあり、T2の板に対して5mmほど掘り込むCAMであったため切削は断念

- [ ] 板厚公差の計測
- [ ] 両面テープの貼付
- [ ] CAMの作成
- [ ] 原点出し
- [ ] 切削

### 2日目

切削できた

- [ ] 板厚公差の計測
- [ ] 両面テープの貼付
- [ ] CAMの作成
- [ ] 原点出し
- [ ] 切削

[image]

RikuYokoo, 2023-04-29:
## 切削日
2023/3/26
## CAM
https://github.com/citbrains/SUSTAINA-OP/commit/123962255ae86549e46ef55dea68db7147f32edb

## 板厚交差
[image]
左下から時計回りに   
-0.07, +0.04, +0.105, +0.09, +0.05, -0.06

## 切削完了
写真を撮り忘れてしまいました．

## エンドミル
over usedのエンドミルを使用
[image]
 ## 加工距離
8.4439m
over usedのため，スプレッドシートには記入していません．
## その他
CAMでは足裏は3個セットされているが，実際の切削は時間の関係上2個しかしてません．

RikuYokoo, 2023-04-29:
## 切削日
2023/4/27, 28
## CAM
https://github.com/citbrains/SUSTAINA-OP/commit/49a1a0fd563f33a3cdaf4775a94bfb11dc04ba49
### 配置図
[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
写真を撮り忘れてしまいました．

【途中です】

MibuchiYuta, 2023-04-30:
# 切削報告
## 切削日
2023/04/29
## CAM
1b0f5f43a79c339fab4c5ab57af9329cef397fcd
## 板厚公差
[image]

全て+0.05mm

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_03
- 加工距離：222.1623
- 総加工距離：222.1623
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-04-30:
## 切削日
2023/4/30
## CAM
ed312e65bff4c41cb8f24517bc7e94d5e1e4669c
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 0.055, 0.055, 0.060, 0.050, 0.045, 0.055

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 04
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  84.4498m
    - 合計
        - 84.4498
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-04-30:
## 切削日
2023/4/30
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 0.062, 0.062, 0.057, 0.06, 0.065, 0.065

## 切削完了
[image]
@MibuchiYuta 
切削完了後写真お願いします．

## エンドミル
- 番号：WXL-3D-DE 04
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  77.3579m
    - 合計
        - 77.3579m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-01:
## 切削日
2023/5/1
## CAM
d2a8d02f3b9964d677ca89c19aec0a20a930c38a
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 0.062, 0.062, 0.057, 0.06, 0.065, 0.065

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 04
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  37.1171m
    - 合計
        - 37.1171m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-01:
## 切削日
2023/5/1
## CAM
d2a8d02f3b9964d677ca89c19aec0a20a930c38a
### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 04
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  37.1171m
    - 合計
        - 37.1171m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-01:
## 切削日
2023/5/1
## CAM
631cbf5d38bac1ad3edd3fbf7e05998385cf5a99
### 配置図

[image]

## 板厚交差

[image]

## 切削完了 (待ち)
[image]

## エンドミル
- 番号：WXL-3D-DE 05
- 加工距離
    - パラメータ遅い
        -  48.8301m
    - パラメータ速い 
        -  36.7993m
    - 合計
        - 85.6294m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/1
## CAM
1b498458c04a80326e924cbbb03c6a44c370ee40
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

次の日に完成するため無し

## エンドミル
- 番号：WXL-3D-DE 05
- 加工距離
    - パラメータ遅い
        -  不明m
    - パラメータ速い 
        -  不明m
    - 合計
        - 不明m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/1
## CAM
1b498458c04a80326e924cbbb03c6a44c370ee40
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

次の日に完成するため無し

## エンドミル
- 番号：WXL-3D-DE 05
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  44.1973m
    - 合計
        - 44.1973m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/2
## CAM
4e9986e3564539316547b0ee786c2de1e1cfa2fe
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 05
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  42.4792m
    - 合計
        - 42.4792m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
# 切削報告(報告忘れ)
## 切削日
2023/04/29
## CAM
1b0f5f43a79c339fab4c5ab57af9329cef397fcd
## 板厚公差
[image]

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_03
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/2
## CAM
a9e80e86e116b4b557e2b6fc1737b1f611abda98
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 06
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  38.9742m
    - 合計
        - 38.9742m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/2
## CAM
8cac6feacc6a7cc6c03754ecb6f34a23c21f871c
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 06
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  35.0416m
    - 合計
        - 35.0416m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-02:
## 切削日
2023/5/3
## CAM
5e222554534e6b9966f4882073bda486ef075326
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 06
- 加工距離
    - パラメータ遅い
        -  12.0001m
    - パラメータ速い 
        -  110.775m
    - 合計
        - 122.7751m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-03:
## 切削日
2023/5/3
## CAM
92d09ee955c38a6d1466d76eb32c730b525dba46
### 配置図

[image]

## 板厚交差

一個前と共通

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 06
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  23.7718m
    - 合計
        - 23.7718m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-03:
## 切削日
2023/5/3
## CAM
ae3e1324651186e46a66c432b0a7fee42de39be3
### 配置図

[image]

## 板厚交差

[image]

## 切削完了

[image]

## エンドミル
- 番号：WXL-3D-DE 06
- 加工距離
    - パラメータ遅い
        -  0m
    - パラメータ速い 
        -  61.2198m
    - 合計
        - 61.2198m
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-04:
## 切削日
2023/05/04
## CAM

### 配置図

[image]

## 板厚交差
[image]

[image]
左下から時計回りに, +0.085, +0.085, +0.085, +0.084, +0.085, +0.085, 

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 07
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  121.903
    - エンドミル総距離
        - 121.903
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-04:
## 切削日
2023/05/04
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 07
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  121.903
    - エンドミル総距離
        - 
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

複数回に分ける

RikuYokoo, 2023-05-05:
## 切削日
2023/
## CAM

### 配置図
[image]

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

2回に分ける

RikuYokoo, 2023-05-06:
## 切削日
2023/05/05
## CAM
8c4271774608ac7c6b2f0abe9f518184c688981f
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-07:
## 切削日
2023/05/06
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-07:
## 切削日
2023/05/06
夜
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-07:
## 切削日
2023/05/07
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_10
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  88.3006
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-08:
## 切削日
2023/05/08
## CAM
@MibuchiYuta 
push後反映お願いします．
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_11
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
@MibuchiYuta 
今回の総距離がわからなかったのと10番がそろそろ300mだったので新しいのを使用しました．
スプレッドシートなどの反映お願いいたします．
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-09:
## 切削日
2023/05/09
## CAM

be3bb06b2f19bcbda6f0508c2dc61b10bc744920

### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_11
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  43.6013
    - 合計
        - 43.6013
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MasatoKubotera, 2023-05-09:
<b> 注：久保寺私物CNC オリジナルマインド SR420での切削記録</b>
## 切削日
2023/05/09

## CAM

d40468b57c0d6382d522f7bbb7ccad12c700fffb

### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：XAL-PEM2R2
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  253.614m
    - 合計
        - 253.614m
  
エンドミル管理対象外

MibuchiYuta, 2023-05-10:
## 切削日
2023/05/09
## CAM

114d1a311184c9e22c250309d57f1b3f5cdfbfd9

### 配置図

[image]
[image]

## 板厚交差

[image]

## 切削完了

失敗した　以降 @RikuYokoo からの報告を待つ

## エンドミル
- 番号：WXL-3D-DE-2_11
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  43.6013
    - 合計
        - 43.6013
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-10:
## 切削日・場所
2023/05/10
新習志野12号館CNC
## CAM
40da69733ad753a5b4a365683b61d564281eb27e
### 配置図
3回に分ける
[image]

## 板厚交差

[image]

左下から時計回りに,

## 切削完了
[image]

完了後報告
## エンドミル
- 番号：WXL-3D-DE-2_11
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-10:
## 切削日・場所
2023/05/09夜
津田沼研究室CNC
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]
撮り忘れ

## エンドミル
- 番号：研究室のCNCのためエンドミルはoveruseまたは，別のを使用のため計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-10:
## 切削日
2023/05/10朝
研究室CNC
## CAM
@MibuchiYuta 
お願いします
### 配置図

[image]

## 板厚交差

[image]
左下から時計回り

## 切削完了
<img src="https://github.com/citbrains/SUSTAINA-OP/assets/53966271/afa5b9ae-86a0-4dfe-954d-216aa3227619" wi
dth=50%>

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-10:
## 切削日・場所
2023/05/09
津田沼研究室CNC
## CAM

### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：研究室のCNCのためエンドミルはoveruseまたは，別のを使用のため計上なし

RikuYokoo, 2023-05-11:
## 切削日
2023/05/10
新習志野12号館
## CAM
A2017T3
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

@MibuchiYuta 
お願いします

RikuYokoo, 2023-05-11:
## 切削日
2023/05/10 夕方
津田沼のCNC
(研究室4回目)
## CAM

### 配置図

## 板厚交差
マイクロメーターを忘れたため、ノギスで計測、信用できない
[image]

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-11:
## 切削日
2023/05/10夜
研究室5回目
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MasatoKubotera, 2023-05-11:
<b> 注：久保寺私物CNC オリジナルマインド SR420での切削記録</b>
## 切削日
2023/05/09

## CAM

d40468b57c0d6382d522f7bbb7ccad12c700fffb

### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：XAL-PEM2R2
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  253.614m
    - 合計
        - 253.614m
 新品エンドミル使用
エンドミル管理対象外

RikuYokoo, 2023-05-11:
## 切削日
2023/05/11
研究室のCNC
## CAM
b38c7438a5a3982e6eb774e2c98733b2d2caf7cf
### 配置図

[image]
[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 14
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  115.6
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-11:
## 切削日・場所
2023/05/11
研究室のCNC
7回目
## CAM
0901.ncを使用
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに

## 切削完了
[image]

## エンドミル
- 番号：管理対象外
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-11:
## 切削日
2023/05/10 
研究室CNC
4回目
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに

## 切削完了
[image]

## エンドミル
管理対象外
- 番号：ロボカップ１
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-11:
## 切削日・場所
2023/05/11
研究室CNC
8回目
## CAM
0901.nc
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：管理対象外
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

## その他
@MibuchiYuta
が担当

RikuYokoo, 2023-05-11:
## 切削日・場所
2023/05/11
研究室CNC
9回目
## CAM
0901.nc
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに,

## 切削完了
[image]

## エンドミル
- 番号：管理対象外
ロボカップ3
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

## その他
@MibuchiYuta が配置などを担当

RikuYokoo, 2023-05-11:
## 切削日・場所
2023/05/11
津田沼研究室
10回目
## CAM
[GFRP_T2_02.iam](https://github.com/citbrains/SUSTAINA-OP/blob/master/Cam/GFRP_T2/OriginalMind_RD400/GFRP_T2_02.iam)

### 配置図
[image]

## 板厚交差

[image]
左下から時計回りに,

## 切削完了
[image]

## エンドミル
- 番号：管理対象外
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-11:
## 切削日・場所
2023/05/11
新習志野12号館
## CAM
4fd2e3f58de4ea496a72a318c2cc48e04a8a5255
[A2017_T2_05.iam](https://github.com/citbrains/SUSTAINA-OP/blob/master/Cam/A2017_T2/A2017_T2_05.iam)
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 14
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  35.3234
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-12:
## 切削日・場所
2023/05/12
研究室CNC
## CAM
[GFRP_T2_02.iam](https://github.com/citbrains/SUSTAINA-OP/blob/master/Cam/GFRP_T2/OriginalMind_RD400/GFRP_T2_02.iam)

### 配置図
[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：エンドミル管理対象外
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MibuchiYuta, 2023-05-12:
## 切削日・場所

新習志野12号館4階工作室
## CAM

c33f2c26ed71fb5da2c1dd6582d6368af858e9ea

### 配置図

[image]

## 板厚交差

[image]

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE-2_15
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  177.764(予定)
    - 合計
        - 177.764(予定)
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/12
研究室CNC12回目
## CAM

### 配置図

[image]

## 板厚交差

[image]

左下から時計回りに, 

## 切削完了

[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/12
研究室CNC
## CAM
0c97da9d1dd83ecd71942b5156fb80d0f85fae49

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/12
研究室CNC
研究室14回目

@MasatoKubotera が担当？
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに,

## 切削完了
[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/12
研究室CNC
@MibuchiYuta が担当？
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/13
研究室CNC
16回目
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに,

## 切削完了
[image]

## エンドミル
- 番号：WXL-3D-DE 16
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  121.903
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/
## CAM

### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/13
新習志野12号館
## CAM
POM
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, -0.07, -0.06, -0.06, -0.06, -0.06, -0.07

## 切削完了
[image]

## エンドミル
- 番号：8番
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/13
新習志野12号館
## CAM

[POM_T5_02.iam](https://github.com/citbrains/SUSTAINA-OP/blob/master/Cam/POM_T5/POM_T5_02.iam)
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：8番
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  67.7437
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

RikuYokoo, 2023-05-13:
## 切削日
2023/05/13
研究室CNC
17回目
## CAM
02
### 配置図

[image]

## 板厚交差

[image]
左下から時計回りに, 

## 切削完了
[image]

## エンドミル
- 番号：計上なし
- 加工距離
    - パラメータ遅い
        -  a
    - パラメータ速い 
        -  a
    - 合計
        - a
  
[エンドミル管理シート](https://docs.google.com/spreadsheets/d/1Jeq6wmjYuJ-2QJXUDLCk-n798fb17JucBtLQSfGDfbc/edit?usp=sharing)

MasatoKubotera, 2023-05-13:
<b> 注：久保寺私物CNC オリジナルマインド SR420での切削記録</b>
## 切削日
2023/05/11

## CAM

59b14cc00cf4fde470d87c134c327f4a097f4adb

### 配置図

[image]

## 板厚交差
[image]

## 切削完了
[image]

## エンドミル
- 番号：XAL-PEM2R2
- 加工距離
    - パラメータ遅い
        -  0
    - パラメータ速い 
        -  227.411m
    - 合計
        - 227.411m
 新品エンドミル使用
エンドミル管理対象外

[31 more replies]

closed 2023-07-25 by MasatoKubotera