SUSTAINA-OP/SUSTAINA-OP #353: SUSTAINA-OP Ver2.0のアルマイト加工の記録

MibuchiYuta, 2023-03-11:


GakuKuwano, 2023-05-04:
# 加工報告

## 加工日
2023/05/03
## 総加工時間
12時間30分

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
58 | 46 | 12 (2) |

## 成功
[image]

## 失敗
[image]

### ボツ
[image]

GakuKuwano, 2023-05-05:
# 加工報告

## 加工日
2023/05/04
## 総加工時間
10時間

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
108 | 89 | 19 (12) |

## 成功
[image]

## 失敗
[image]

### ボツ
[image]

GakuKuwano, 2023-05-06:
# 加工報告

## 加工日
2023/05/05
## 総加工時間
12時間

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
131 | 98 | 33 (脱色25,ボツ7) |

## 成功
[image]

## 失敗
- 染色液，封孔液の交換のタイミングがわからなくてミス（脱色）が増えた
[image]

### 脱色
[image]

### ボツ
[image]

GakuKuwano, 2023-05-07:
# 加工報告
染色液を交換した
## 加工日
2023/05/06
## 総加工時間
5時間

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
33 | 30| 3 (3) |

## 成功
[image]

## 失敗
[image]

GakuKuwano, 2023-05-10:
# 加工報告
封孔液を交換した
## 加工日
2023/05/10
## 総加工時間
5時間

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
54 | 53 | 1 (1) |

## 成功
[image]

## 失敗
### ボツ
[image]

GakuKuwano, 2023-05-12:
# 加工報告
染色液を交換した
## 加工日
2023/05/11
## 総加工時間
7時間

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
88 | 84 | 4 (4) |

## 成功
[image]

## 失敗
### ボツ
[image]

GakuKuwano, 2023-05-14:
# 加工報告
染色液，電解液を交換した
## 加工日
2023/05/12

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
58 | 50 | 8 (8) |

## 成功
[image]

## 失敗
### ボツ
[image]

GakuKuwano, 2023-05-15:
# 加工報告
## 加工日
2023/05/14

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
89 | 78 | 11 (10) |

## 成功
[image]

## 失敗
### ボツ
[image]

### 脱色
[image]

GakuKuwano, 2023-05-15:
# 加工報告
## 加工日
2023/05/15

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
54 | 54 | 0 |

## 成功
[image]

GakuKuwano, 2023-05-21:
# 加工報告
## 加工日
2023/05/19

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
170 | 149 | 21 |

## 成功
[image]
[image]

##失敗
[image]

GakuKuwano, 2023-05-21:
# 加工報告
## 加工日
2023/05/20

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
56 | 53 | 3 |

## 成功
[image]

## 失敗
[image]

GakuKuwano, 2023-05-24:
# 加工報告
## 加工日
2023/05/22

## 結果
総数 | 成功 | 失敗 (ボツ) |
:---: | :---: | :---:
73 | 68 | 5 |

## 成功
[image]

## 失敗
[image]

closed 2023-07-25 by MasatoKubotera