SUSTAINA-OP/SUSTAINA-OP #355: 動作時に不定期でサーボモータと通信するUSBデバイスがロストする

hiroki-0001, 2023-07-27:
RoboCup2023期間中にも発生していた問題。主に機体番号183で発生する。
症状に関してはissue名通りで、ロボット動作中にサーボモータと通信するデバイスが`lsusb`コマンドで確認できなくなり、サーボモータを制御することができなくなる。

`dmesg`等でカーネルの出力を確認すると以下のようなエラーが発生している。

```
Aug 18 13:24:07 [kernel] ftdi_sio ttyUSB2: error from flowcontrol urb
Aug 18 13:24:12 [kernel] ftdi_sio ttyUSB2: ftdi_set_termios urb failed to set baudrate
```

MasatoKubotera, 2023-07-28:
<code>lsusb</code>で確認できなくなっているため，再接続すらできない状況に陥っている．

問題が生じているモジュール：
https://github.com/SUSTAINA-OP/USB-to-Quad-RS-485-Conv-Module

上記のモジュールは，RoboCup2022まで使用していたFT4232H MINI MODULEとUART/RS-485 変換を併せたカスタムモジュール．
EEPROMはデフォルト構成からの変更点は，RS-485を有効化するために，```<RI_RS485>true</RI_RS485>```としているのみである．

FT3242H MINI MODULEのデフォルト構成とFT4232HQ（モジュール上のチップ）のデフォルト構成を以下にまとめた．
https://github.com/SUSTAINA-OP/FT4232H_EEPROM_Config

ここで，以下の2点が相違があることに気づく．
```.diff
-     <RemoteWakeupEnabled>false</RemoteWakeupEnabled>
+     <RemoteWakeupEnabled>true</RemoteWakeupEnabled>
```
```.diff
-   <MaxPower>100</MaxPower>
+   <MaxPower>150</MaxPower>
```

よってこの2点の設定によって，この問題が起きている可能性がわかってきた．

MasatoKubotera, 2023-07-28:
参考：bitbots_quaddxlの構成
https://github.com/bit-bots/bitbots_quaddxl/blob/master/firmware/main.c

> Data.MaxPower = 44;
> Data.RemoteWakeup = 1;

MasatoKubotera, 2023-07-28:
IO電流も異なっていたが，現状搭載しているUART/RS-485については，デフォルトの4mAで問題ないよう．

```diff
...
  <Group_A>
    <SlowSlew>false</SlowSlew>
    <Schmitt>false</Schmitt>
-     <Drive>4mA</Drive>
+     <Drive>8mA</Drive>
  </Group_A>
...
```

Input Current (A, B) IA, IB Max.125 µA (DE = 0 V, VCC = 0 V or 3.6 V, VIN = 12 V)
https://www.analog.com/media/en/technical-documentation/data-sheets/ADM3070E_3071E_3072E_3073E_3074E_3075E_3076E_3077E_3078E.pdf#page=4

MasatoKubotera, 2023-07-28:
A203に乗っているUSB-HUB ICを調査した．

RTS5411 
https://www.realtek.com/ja/products/computer-peripheral-ics/category/usb3-2-gen1-hubs

該当のFT4232Hは，A203上のRTS5411のダウンストリームポート2と，拡張基板のUSB-HUBを経由しており，
このどちらとも，USBのバスパワーを制御するＩＣ等に接続していないことが分かった．
（しかし，A203のGPIOピンの電源5Vとは別系統）

そのため， MaxPowerによって電流制限がおこなわれていることはなさそう．
（示しているだけで，物理的な処理をしている/するわけではなさそう．）

MasatoKubotera, 2023-07-28:
よって，今回の問題は，RemoteWakeupEnabledの設定によるものである可能性が高い．

しかし，RemoteWakeupEnabledが関与するのは，サスペンドモードになることである．

競技中にサスペンドになる＝FT4232Hを使用していない＝アクチュエータを制御していない時がある

@AD58-3104 によると，戦略を投げてから制御しないことはないとのことなのでどこかしらにバグがある可能性はある．

MasatoKubotera, 2023-07-28:
サスペンドからの復帰には20msかかるようです．
https://ftdichip.com/wp-content/uploads/2020/08/DS_FT4232H.pdf#page=35