SUSTAINA-OP/SUSTAINA-OP #356: Jetson経由でFT4232HのEEPROMを読み書きできるようにする

MasatoKubotera, 2023-07-28:
アクチュエータを駆動するためのデバイス[USB to Quad RS-485 Conv. Module](https://github.com/SUSTAINA-OP/USB-to-Quad-RS-485-Conv-Module)のEEPROM構成はデフォルトから書き直す必要がある．

これまでは，[FT_PROG](https://ftdichip.com/utilities/)を用いてWindowsから書き込む必要があった．
しかし，書き込むためにはUSBケーブルを挿抜しなければならず大変であるため，Jetson経由のUbuntu OSで書き込めるようにしたい．

参考：https://github.com/bit-bots/bitbots_quaddxl/tree/master/firmware

MasatoKubotera, 2023-07-28:
EEPROM構成は以下の通り．

https://github.com/SUSTAINA-OP/FT4232H_EEPROM_Config/blob/master/USB-to-Quad-RS-485-Conv-Module_custom_configuration.xml

AD58-3104, 2023-08-16:
ftd2xx（C言語のライブラリ）からEEPROMに書き込む時のデータの名前と、FT_progから書き込むデータの名前に少し齟齬があり、よく分からないので確認したい。

今の所の、ftd2xxでデータを書き込む際の設定は以下。

``` 
	FT_PROGRAM_DATA Write_Data = {
		0x00000000,				   // Header - must be 0x00000000
		0xFFFFFFFF,				   // Header - must be 0xffffffff
		0x00000004,				   // Header - FT_PROGRAM_DATA version
		0x0403,					   // VID
		0x6011,					   // PID
		"FTDI",					   // Manufacturer
		"FT",					   // Manufacturer ID
		"USB <-> Serial Converter", // Description
		"FT000001",				   // Serial Number     TODO::ここ不明
		100,					   // MaxPower
		1,						   // PnP
		0,						   // SelfPowered
		1,						   // RemoteWakeup
		1,						   // non-zero if Rev4 chip, zero otherwise TODO::ここ不明
		0,						   // non-zero if in endpoint is isochronous TODO::ここ不明
		0,						   // non-zero if out endpoint is isochronous TODO::ここ不明
		0,						   // non-zero if pull down enabled TODO::ここ不明
		1,						   // non-zero if serial number to be used TODO::ここ不明
		0,						   // non-zero if chip uses USBVersion TODO::ここ不明
		0x0200,					   // BCD (0x0200 => USB2)
		//
		// FT2232C extensions (Enabled if Version = 1 or greater)
		//
		0,	 // non-zero if Rev5 chip, zero otherwise
		0,	 // non-zero if in endpoint is isochronous
		0,	 // non-zero if in endpoint is isochronous
		0,	 // non-zero if out endpoint is isochronous
		0,	 // non-zero if out endpoint is isochronous
		0,	 // non-zero if pull down enabled
		0,	 // non-zero if serial number to be used
		0,	 // non-zero if chip uses USBVersion
		0x0, // BCD (0x0200 => USB2)
		0,	 // non-zero if interface is high current
		0,	 // non-zero if interface is high current
		0,	 // non-zero if interface is 245 FIFO
		0,	 // non-zero if interface is 245 FIFO CPU target
		0,	 // non-zero if interface is Fast serial
		0,	 // non-zero if interface is to use VCP drivers
		0,	 // non-zero if interface is 245 FIFO
		0,	 // non-zero if interface is 245 FIFO CPU target
		0,	 // non-zero if interface is Fast serial
		0,	 // non-zero if interface is to use VCP drivers
		//
		// FT232R extensions (Enabled if Version = 2 or greater)
		//
		0, // Use External Oscillator
		0, // High Drive I/Os
		0, // Endpoint size
		0, // non-zero if pull down enabled
		0, // non-zero if serial number to be used
		0, // non-zero if invert TXD
		0, // non-zero if invert RXD
		0, // non-zero if invert RTS
		0, // non-zero if invert CTS
		0, // non-zero if invert DTR
		0, // non-zero if invert DSR
		0, // non-zero if invert DCD
		0, // non-zero if invert RI
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // non-zero if using D2XX drivers
[... 72 lines cut ...]
		0, // Cbus Mux control
		0, // non-zero if interface is 245 FIFO
		0, // non-zero if interface is 245 FIFO CPU target
		0, // non-zero if interface is Fast serial
		0, // non-zero if interface is FT1248
		0, // FT1248 clock polarity - clock idle high (1) or
		// clock idle low (0)
		0, // FT1248 data is LSB (1) or MSB (0)
		0, // FT1248 flow control enable
		0, // non-zero if interface is to use VCP drivers
		0 // non-zero if using ACBUS7 to save power for
		// self-powered designs
	};
```

今の所分からない点は、以下

- プルダウン有効化の設定が、FT4232H用と共通の設定で2つある。
- xmlでは、Serial Numberを自動で振る設定になっているが、ftd2xxの方にはそのような設定が無い。
- xmlにはBusPoweredの設定があるが、ftd2xxにはそのような設定が無い。
- xmlにはUSBVersionの設定が無い。
- xmlにはRev4 chipを使っているか否かの設定が無い。
- xmlにはPnPの設定が無い。
- xmlにはin endpoint is isochronousの設定が無い
- xmlにはout endpoint is isochronousの設定が無い

MasatoKubotera, 2023-08-16:
>xmlでは、Serial Numberを自動で振る設定になっているが、ftd2xxの方にはそのような設定が無い。

シリアルナンバは処理的に使用していないため，ランダムもしくは無効．被っても問題はない．

> xmlにはBusPoweredの設定があるが、ftd2xxにはそのような設定が無い。

WORD SelfPowered;			// 0 = bus powered, 1 = self powered

>xmlにはRev4 chipを使っているか否かの設定が無い。

Rev4 (FT232B) extensionsのみの設定だと思われる

AD58-3104, 2023-08-16:
ちなみにソフトウェアのリファレンスはこれ。
　https://www.ftdichip.com/Support/Documents/ProgramGuides/D2XX_Programmer's_Guide(FT_000071).pdf

MasatoKubotera, 2023-08-16:
@AD58-3104 
ソースコード類は，以下のリポジトリの直下にfirmwareディレクトリ作ってその中に格納してほしい
https://github.com/SUSTAINA-OP/USB-to-Quad-RS-485-Conv-Module

MasatoKubotera, 2023-08-16:
FT4232Hのデータシート
https://ftdichip.com/wp-content/uploads/2020/08/DS_FT4232H.pdf

AD58-3104, 2023-08-22:
最終的にftd2xxを利用して書き込んだ設定は以下。
```
//書き込みデータの設定-----------------------------------------------------------
	FT_PROGRAM_DATA Write_Data = {
		0x00000000,				   // Header - must be 0x00000000
		0xFFFFFFFF,				   // Header - must be 0xffffffff
		0x00000004,				   // Header - FT_PROGRAM_DATA version
		0x0403,					   // VID
		0x6011,					   // PID
		"FTDI",					   // Manufacturer
		"FT",					   // Manufacturer ID
		"USB <-> Serial Converter", // Description
		"FT000001",				   // Serial Number     シリアルナンバーは適当で良いらしいので、全て同じこの値にした。
		100,					   // MaxPower
		1,						   // PnP      FT232Bのextensionのみに関連する設定と思われる
		0,						   // SelfPowered
		1,						   // RemoteWakeup
		1,						   // non-zero if Rev4 chip, zero otherwise FT232Bのみの設定と思われるが、取りあえずオン
		0,						   // non-zero if in endpoint is isochronous FT232Bのみの設定と思われるが、取りあえずオフ
		0,						   // non-zero if out endpoint is isochronous FT232Bのみの設定と思われるが、取りあえずオフ
		0,						   // non-zero if pull down enabled ここは無効とする
		1,						   // non-zero if serial number to be used 
		0,						   // non-zero if chip uses USBVersion ここはfalseにして良いらしい。
		0x0200,					   // BCD (0x0200 => USB2)
		//
		// FT2232C extensions (Enabled if Version = 1 or greater)
		//
		0,	 // non-zero if Rev5 chip, zero otherwise
		0,	 // non-zero if in endpoint is isochronous
		0,	 // non-zero if in endpoint is isochronous
		0,	 // non-zero if out endpoint is isochronous
		0,	 // non-zero if out endpoint is isochronous
		0,	 // non-zero if pull down enabled
		0,	 // non-zero if serial number to be used
		0,	 // non-zero if chip uses USBVersion
		0x0, // BCD (0x0200 => USB2)
		0,	 // non-zero if interface is high current
		0,	 // non-zero if interface is high current
		0,	 // non-zero if interface is 245 FIFO
		0,	 // non-zero if interface is 245 FIFO CPU target
		0,	 // non-zero if interface is Fast serial
		0,	 // non-zero if interface is to use VCP drivers
		0,	 // non-zero if interface is 245 FIFO
		0,	 // non-zero if interface is 245 FIFO CPU target
		0,	 // non-zero if interface is Fast serial
		0,	 // non-zero if interface is to use VCP drivers
		//
		// FT232R extensions (Enabled if Version = 2 or greater)
		//
		0, // Use External Oscillator
		0, // High Drive I/Os
		0, // Endpoint size
		0, // non-zero if pull down enabled
		0, // non-zero if serial number to be used
		0, // non-zero if invert TXD
		0, // non-zero if invert RXD
		0, // non-zero if invert RTS
		0, // non-zero if invert CTS
		0, // non-zero if invert DTR
		0, // non-zero if invert DSR
		0, // non-zero if invert DCD
		0, // non-zero if invert RI
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // non-zero if using D2XX drivers
		//
		// Rev 7 (FT2232H) Extensions (Enabled if Version = 3 or greater)
[... 59 lines cut ...]
		0, // non-zero if AD pins are Schmitt input
		0, // valid values are 4mA, 8mA, 12mA, 16mA
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // Cbus Mux control
		0, // non-zero if interface is 245 FIFO
		0, // non-zero if interface is 245 FIFO CPU target
		0, // non-zero if interface is Fast serial
		0, // non-zero if interface is FT1248
		0, // FT1248 clock polarity - clock idle high (1) or
		// clock idle low (0)
		0, // FT1248 data is LSB (1) or MSB (0)
		0, // FT1248 flow control enable
		0, // non-zero if interface is to use VCP drivers
		0 // non-zero if using ACBUS7 to save power for
		// self-powered designs
	};
```

AD58-3104, 2023-08-22:
@MasatoKubotera 
リポジトリにソースコードをアップロードしようと思ったんだけど、権限がないみたい...。

MasatoKubotera, 2023-08-22:
@AD58-3104 対応しました．