SUSTAINA-OP/SUSTAINA-OP #357: 機体番号183のB3Mサーボモータが動作が不安定

hiroki-0001, 2023-07-28:
機体番号183で発生する問題である。以下のような症状がロボットの全関節(19個分)で発生

- B3Mの状態を示すLEDが点灯していない
- B3Mの出力軸の動きが硬い(ホールドモードに近い？)
- B3M Manegerを利用した通信不可能
    - 事前に書き込んだIDを指定して通信を行った。つまりbroadcast ID 255は不使用。これはデイジーチェーン中のB3MのIDの混乱を防ぐため

問題が発生した時の動画：https://youtu.be/qc-kdtBFnBU

hiroki-0001, 2023-07-28:
B3M側が今まで観測したことのない異常な症状なので、https://kondo-robot.com/contacts に一度報告してみます。

MasatoKubotera, 2023-07-29:
7月28日にスイッチング電源を用いた時に186機でも発生している．

MasatoKubotera, 2023-08-16:
原因

- P-ch FETを用いた突入電流防止回路によるスロースタート
- 大容量コンデンサ 150uF × 4個の搭載

hiroki-0001, 2023-08-16:
近藤科学から回答をいただきました。以下のドキュメントにまとめておきます。

https://docs.google.com/document/d/1ASpvD2zdjgcxrFOtTs4T46jRfEMR--8DcKEUh9-CHbw/edit?usp=sharing

hiroki-0001, 2023-08-31:
https://github.com/SUSTAINA-OP/SUSTAINA-OP/issues/357#issue-1826196872
と同じような状態が**バッテリー駆動時の182**で発生しました．

ロボット背面の電圧表示が11.2V程度の際にサーボモータの電源スイッチをONにした際， Jetsonの電源に落ち，B3Mが上記のリンクのような症状になりました．

その時の動画：https://youtu.be/wT_9sl5OkyA?t=1790
INA226による電圧・電流計測値：https://github.com/citbrains/sustaina-log/commit/2f04e6a866bc8e5144a34d00f89784de7bd1a3b1#diff-f6642c02da68a80e069748ef0b5a44e5442599a8ef6f71dcea341c625bcaf935