SUSTAINA-OP/SUSTAINA-OP #360: 歩行中に電源が落ちる

MasatoKubotera, 2023-08-16:
230816

185機を用いて以下の検証した際，歩行中に電源が落ちる場面があった．
https://github.com/SUSTAINA-OP/SUSTAINA-OP/issues/356

Jetson の電源が落ち，アクチュエータが脱力したほか，電圧計の表示が不灯，コックピット内からシリコン系の異臭がした．

アクチュエータのスイッチをOFFにすることで，電圧計の表示は通常になった．

以下，定点カメラのアーカイブ
https://youtu.be/ektLZj52BO4
https://youtu.be/LE7SSuMGEoE

MasatoKubotera, 2023-08-16:
INAによる電圧電流ログは，ソフトウェア側の欠陥により，取得できていなかった．

MasatoKubotera, 2023-08-16:
シングルボードコンピュータユニット（Jetson + CarrierBoard）単体をACアダプタ電源経由で動作確認できた．

デバイスはすべて認識したようだ．
※アクチュエータの駆動はしていない．

MasatoKubotera, 2023-08-16:
[SUSTAINA-Core-Board](https://github.com/SUSTAINA-OP/SUSTAINA-Core-Board)周辺回路のみでスイッチング電源に接続して，優先1位，2位供給口から電源を投入したが，ヒューズ等が切れていることは確認できなかった．
また，通電状態を示すLEDの点灯についてもすべて転倒しており問題は確認できなかった．

SBC電源出力についても既定値の出力が確認できている．

[image]

MasatoKubotera, 2023-08-16:
メモ

電力計測モジュールとシングルボードコンピュータを直接繋ぐのは，リスクマネジメントとして避けた方が良いかもしれない．
また，RS-485系統も同等である．

MasatoKubotera, 2023-08-26:
歩行中に電源が落ちた原因が特定できた

首Yaw軸のサーボモータ内部基板のFET（SP8M4）1個が焼けていることが確認できた．
これにより，サーボモータ電源が入らない状況だった．

コックピット内部から異臭がしたのは，該当のサーボモータの直下にコンピュータユニットのファンが設置されていると推測される．

[image]
[image]
[image]

MasatoKubotera, 2023-08-26:
動画を確認したが，首のモータに干渉が起きていることは確認できなかった．

MasatoKubotera, 2023-08-26:
今回の問題点としては，モータ電源のヒューズ（２０A）が作動しなかったこと．
基板の仕様的に２０A流れなかった可能性がある．
これにより，SBC電源まで制限されコンピュータ電源が落ちたと考えられる．

hiroki-0001, 2023-09-06:
近藤科学さんの方に今回の問題について問い合わせました。
やり取り・回答に関しては以下のドキュメントを参照してください。