SUSTAINA-OP/SUSTAINA-OP #362: 通信デバイスを一系統のUSBバスで完結させる

MasatoKubotera, 2023-08-29:
MIPI SCI接続のカメラモジュール以外の通信デバイス（INA，IMU，USB/RS-485，制御ボタン）をUSB-HUBを介した一系統のUSBバスで完結させる．

これにより，シングルボードコンピュータの環境に依存しない回路構成としたい．

yasuohayashibara, 2023-08-29:
経験的にUSB-HUBを介して複数の機器の通信を行うと，通信エラーが多発することがあるので気をつけて下さい．

MasatoKubotera, 2023-08-29:
メモ：つくばチャレンジでもUSB-HUB関連の問題が挙げられている．（10年ほど前から）

MasatoKubotera, 2023-08-30:
現状，USB/RS-485変換に利用しているFT4232HがUSB/SPI，USB/I2Cにも対応している．
ロングセラーのチップなので，扱いやすいかも
https://ftdichip.com/Documents/AppNotes/AN_113_FTDI_Hi_Speed_USB_To_I2C_Example.pdf

MasatoKubotera, 2023-10-13:
FT4232H MINI MODULEと2台のXAIO(Arduino)をUART経由（ArduinoはSerial1を使用）で送受信できるか確かめてみた．

FT4232Hから特定の文字を送信して，XAIOが文字を受信したら文字列（float型をbyte型に変換した文字列）を送り返す．
文字列は受信ズレが起こるのでヘッダフッターを入れてあげれば，実用できそうな感じであることを確認．

[image]

[image]

```.py
import serial
import time
import struct

# FT4232Hのシリアルポート設定
ser = serial.Serial('/dev/ttyUSB0', 115200)  # 'COM1'はFT4232Hのポートに置き換えてください
ser.timeout = 1  # タイムアウトの設定

try:
    while True:
        # 1と0の文字列を交互に送信
        ser.write(b'1\n')

        received_data = ser.read(32)

        if len(received_data) == 32:
            try:
                float_values = struct.unpack('8f', received_data)
                print("Received :", float_values)
            except struct.error:
                print("Failed to unpack received data")

            time.sleep(1/10)

except KeyboardInterrupt:
    print("プログラムが停止しました。")

finally:
    ser.close()
```

```.py
import serial
import time
import struct

# FT4232Hのシリアルポート設定
ser = serial.Serial('/dev/ttyUSB1', 115200)  # 'COM1'はFT4232Hのポートに置き換えてください
ser.timeout = 1  # タイムアウトの設定

try:
    while True:
        # 1と0の文字列を交互に送信
        ser.write(b'1\n')

        received_data = ser.read(28)

        if len(received_data) == 28:
            try:
                float_values = struct.unpack('7f', received_data)
                print("Received :", float_values)
            except struct.error:
                print("Failed to unpack received data")

            time.sleep(1/10)

except KeyboardInterrupt:
    print("プログラムが停止しました。")

finally:
    ser.close()
```

```.ino
void setup()
{
  // serial to display data
  Serial1.begin(115200);
}

void loop()
{
  if (Serial1.available())
  {
    byte received_char = Serial1.read();

    float a_mA = 10.00;
    float a_mV = 1000.00;
    float b_mA = 11.00;
    float b_mV = 1100.00;
    float c_mA = 12.00;
    float c_mV = 1200.00;
    float d_mA = 13.00;
    float d_mV = 1300.00;

    // convert 6 float variables to byte array
    byte *ptr = NULL;
    byte data[sizeof(float) * 8];
    for (int data_itr = 0; data_itr < 8; data_itr++)
    {
      switch (data_itr)
      {
        case 0:
          ptr = (byte *)&a_mA;
          break;
        case 1:
          ptr = (byte *)&a_mV;
          break;
        case 2:
          ptr = (byte *)&b_mA;
          break;
        case 3:
          ptr = (byte *)&b_mV;
          break;
        case 4:
          ptr = (byte *)&c_mA;
          break;
        case 5:
          ptr = (byte *)&c_mV;
          break;
        case 6:
          ptr = (byte *)&d_mA;
          break;
        case 7:
          ptr = (byte *)&d_mV;
          break;
      }
      for (int ptr_itr = 0; ptr_itr < sizeof(float); ptr_itr++)
      {
        data[data_itr * sizeof(float) + ptr_itr] = *ptr;
        ptr++;
      }
    }

    // send byte array via serial communication
    Serial1.write(data, sizeof(data));
  }
}
```

```.ino
void setup()
{
  // serial to display data
  Serial1.begin(115200);
}

void loop()
{
  if (Serial1.available())
  {
    byte received_char = Serial1.read();

    float a_x = 1234.0;
    float a_y = 1234.0;
    float a_z = 1234.0;
    float g_x = 1234.0;
    float g_y = 1234.0;
    float g_z = 1234.0;
    float temp_ = 1234.0;

    // convert 6 float variables to byte array
    byte *ptr = NULL;
    byte data[sizeof(float) * 7];
    for (int data_itr = 0; data_itr < 7; data_itr++)
    {
      switch (data_itr)
      {
        case 0:
          ptr = (byte *)&a_x;
          break;
        case 1:
          ptr = (byte *)&a_y;
          break;
        case 2:
          ptr = (byte *)&a_z;
          break;
        case 3:
          ptr = (byte *)&g_x;
          break;
        case 4:
          ptr = (byte *)&g_y;
          break;
        case 5:
          ptr = (byte *)&g_z;
          break;
        case 6:
          ptr = (byte *)&temp_;
          break;
      }
      for (int ptr_itr = 0; ptr_itr < sizeof(float); ptr_itr++)
      {
        data[data_itr * sizeof(float) + ptr_itr] = *ptr;
        ptr++;
      }
    }

    // send byte array via serial communication
    Serial1.write(data, sizeof(data));
  }
}
```